ioBroker/ioBroker.modbus #450: How to change Growatt holding register which are holding times?

pingopilot, 2024-01-29:
I'm not sure if it is a bug, but if I change a Growatt holding register (which holds the timecode hhmm) with the Modbus adapter with the Javascript setState command, I get weird times as a result. Probably I have to write in a different format?

In the Growatt register description it says: _
The format for each byte in RTU mode is:
Coding System: 8–bit binary, hexadecimal 0–9, A–F
Two hexadecimal characters contained in each
8–bit field of the message_

[image]

I read out this register with this option:
[image]

and then I get these objects in iobroker:
[image]

Now I can read the right times in minutes and hour. So far so good. Now I want to change Now I want to change these times by writing in the objects:

But if I write then a different number with the javascript command setState to this object to change the time, it is not changing the time right. First I see my choosen number in the object, but then it changes to a weird different time (changing minutes works but no the hours). That means in my view, that I write the wrong number to the register. I played with different Types but I don't understand what is happening there.
How is the Modbus adapter writing in the register, when I change one object? 
Which format I have to use to get the right result?

Thanks a lot for your help

PLCHome, 2024-01-31:
256*hour + minute is actually correct or (hour << 8) + minute
Go back with us
minutes = (value & 255)
Hour = (value >> 8)
With 16 bit big endian and with 16 bit little endian the hour and minute are swapped.

I'm guessing that the missing ack irritates him.

`setState('modbus.0.holdingRegisters.41101_Bat_First_Start_Time_1_hour'/41101 Bat First Start Time 1 Hour/,10, false)`

PLCHome, 2024-02-01:
<< and >> are commands, more precisely, arithmetic operations

<< 8 shifts the number 8 bits to the left:
so 1 in binary:
0000 0000 0000 0001
gives 1 << 8 then
0000 0001 0000 0000
gives 256

to the other side:
>> 8 shifts the number 8 bits to the right:
258
0000 0001 0000 0010
results in 258 >> 8
0000 0000 0000 0001
results in 1. The bit 2^1 is simply "moved away"

https://www-user.tu-chemnitz.de/~heha/hsn/chm/javascript.chm/04-AUO.htm
https://www.w3schools.com/js/js_bitwise.asp

PLCHome, 2024-02-01:
If your problem is solved, the community will be happy if you close the thread. 

Otherwise feel free to ask more :-)

pingopilot, 2024-02-01:
Great help. Thanks

closed 2024-02-01 by pingopilot