ioBroker/ioBroker.modbus #459: no function of modbus1 in iobroker after update

LCNJoe, 2024-02-04:
After the update to 6.1.0, no data is available from the modbus1 instance. Poll error account .. Restart adapter system.adapter.modbus.1 because enabled.

Modbus0 to SMA inverter continues to work well. Modbus1 queries a smartmeter SDM630 via a PE11 converter (TCP-modbus). The wallbox continues to query the smart meter and data is processed.

What has changed in the new version? What can I adjust?

Thanks for your help

Apollon77, 2024-02-04:
Please provide a debug log from the not working instance. thank you

LCNJoe, 2024-02-04:
ok, here is the debug log from just now, I set the level to debug shortly before.

[Debud_modbus1_Instance_20240204.txt](https://github.com/ioBroker/ioBroker.modbus/files/14157132/Debud_modbus1_Instance_20240204.txt)

Apollon77, 2024-02-04:
Hm ... in fact the adapter can not connect/get no data from device and so restarts itself after some tries to retryx cleanly

LCNJoe, 2024-02-04:
It worked for a year with the previous version. Therefore, something in the adapter must have been changed in the current version. A new variable or library perhaps?

Apollon77, 2024-02-04:
Did you tried reinstalling the old version to verify it is aboit the version and not about the device?

LCNJoe, 2024-02-05:
I can do it, how can I install the old version? And which device do you mean? The TCP modbus device works perfectly, it is queried correctly by my wallbox.

Apollon77, 2024-02-05:
Admin - expert mode - adapters - there is an icon right where you get a list of previous versions and can choose one to install

LCNJoe, 2024-02-05:
OK, I already found it, thanks.
the previous version is running. I've attached the log, maybe it will help with troubleshooting.

[Github_issue_459_modbus_log_2.txt](https://github.com/ioBroker/ioBroker.modbus/files/14162691/Github_issue_459_modbus_log_2.txt)