ioBroker/ioBroker.modbus #472: ReadInputRegisters: Invalid FC 132 with Sungrow inverter

mlankenau, 2024-03-14:
**Describe the bug**  
For a week my IOBroker instance is sometimes failing to receive the data from my sungrow inverter and logging "Invalid FC 132 with Sungrow inverter"

**To Reproduce**  
Steps to reproduce the behavior:
1.) Setup Modbus for the sugrow inverter as described above
2.) wait for the errors

**Expected behavior**  
No errors, steady stream of measurements

**Screenshots & Logfiles**  

I receive:

2024-03-14 09:30:48.520	error	Client in error state.
2024-03-14 09:30:48.519	error	Request timed out.
2024-03-14 09:30:48.518	warn	Error: undefined
2024-03-14 09:30:16.505	warn	Poll error count: 2 code: "App Timeout"
2024-03-14 09:29:46.311	warn	Error in handler for 4: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access 2024-03-14 09:29:44.293	warn	Poll error count: 1 code: "ReadInputRegisters: Invalid FC 132"

My Register setup:
[image]

My general setup:
[image]

**Versions:**  
 - Adapter version: 6.1.0 and 6.0.0
 - JS-Controller version: 4.0.23
 - Node version: v16.18.1
 - Operating system: docker-image: buanet/iobroker on Ubuntu

**Additional context**  

It is not allways happening. If I restart the adapter it works for 30/60minuntes, then the errors start, but I sometime still get updates.

smartcuc, 2024-04-06:
Well, since last update I'm having the same issue. Will try to downgrade the JS-Contr. for verification.

larspoeggeler, 2024-04-30:
Hi,

I have the same issues. 

Adapter v6.1.0
Node.js v18.19.0
JS-Controller v5.0.19

`

2024-04-30 01:21:24.668 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:21:24.684 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:21:24.685 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:21:24.688 - warn: modbus.0 (5543) Poll error count: 1 code: {"err":"timeout","timeout":5000}
2024-04-30 01:21:25.685 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:21:27.692 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:21:29.244 - warn: modbus.0 (5543) Error in handler for 3: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-04-30 01:21:32.895 - warn: modbus.0 (5543) Poll error count: 2 code: "App Timeout"
2024-04-30 01:21:38.949 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:21:38.951 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:21:38.952 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:21:38.952 - warn: modbus.0 (5543) Poll error count: 3 code: {"err":"timeout","timeout":5000}
2024-04-30 01:21:39.952 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:21:41.957 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:21:41.969 - warn: modbus.0 (5543) Poll error count: 4 code: "ReadInputRegisters: Invalid FC 0"
2024-04-30 01:21:48.024 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:21:48.025 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:21:48.026 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:21:48.028 - warn: modbus.0 (5543) Poll error count: 5 code: {"err":"timeout","timeout":5000}
2024-04-30 01:21:49.026 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:21:51.033 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:21:56.037 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:21:56.038 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:21:56.039 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:21:56.039 - warn: modbus.0 (5543) Poll error count: 6 code: {"err":"timeout","timeout":5000}
2024-04-30 01:21:57.039 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:21:59.045 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:04.048 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:04.049 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:04.050 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:22:04.051 - warn: modbus.0 (5543) Poll error count: 7 code: {"err":"timeout","timeout":5000}
2024-04-30 01:22:05.050 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:22:07.056 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:12.058 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:12.059 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:12.059 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:22:12.060 - warn: modbus.0 (5543) Poll error count: 8 code: {"err":"timeout","timeout":5000}
2024-04-30 01:22:13.061 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:22:15.067 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:20.069 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:20.070 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:20.071 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:22:20.072 - warn: modbus.0 (5543) Poll error count: 9 code: {"err":"timeout","timeout":5000}
2024-04-30 01:22:21.071 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:22:23.076 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:28.079 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:28.080 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:28.080 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:22:28.081 - warn: modbus.0 (5543) Poll error count: 10 code: {"err":"timeout","timeout":5000}
2024-04-30 01:22:29.081 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:22:31.087 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:36.090 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:36.091 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:36.092 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:22:36.092 - warn: modbus.0 (5543) Poll error count: 11 code: {"err":"timeout","timeout":5000}
2024-04-30 01:22:37.092 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:22:39.099 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:44.102 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:44.103 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:44.103 - error: modbus.0 (5543) Client in error state.
2024-04-30 01:22:44.105 - warn: modbus.0 (5543) Poll error count: 12 code: {"err":"timeout","timeout":5000}
2024-04-30 01:22:45.104 - info: modbus.0 (5543) Disconnected from slave 192.168.18.65
2024-04-30 01:22:47.110 - info: modbus.0 (5543) Connected to slave 192.168.18.65
2024-04-30 01:22:52.112 - warn: modbus.0 (5543) Error: undefined
2024-04-30 01:22:52.114 - error: modbus.0 (5543) Request timed out.
2024-04-30 01:22:52.114 - error: modbus.0 (5543) Client in error state.
[... cut ...]
2024-04-30 01:22:52.903 - info: host.ioBroker instance system.adapter.modbus.0 terminated with code 156 (START_IMMEDIATELY_AFTER_STOP)
2024-04-30 01:22:52.924 - info: host.ioBroker Restart adapter system.adapter.modbus.0 because enabled
2024-04-30 01:22:54.180 - info: host.ioBroker instance system.adapter.modbus.0 started with pid 5604
2024-04-30 01:22:58.577 - info: modbus.0 (5604) starting. Version 6.1.0 in /opt/iobroker/node_modules/iobroker.modbus, node: v18.19.0, js-controller: 5.0.19
2024-04-30 01:23:01.365 - info: modbus.0 (5604) Connected to slave 192.168.18.65

`