ioBroker/ioBroker.modbus #500: Warning regarding scale factor after upgrade to 6.2.0

disaster123, 2024-04-12:
**Describe the bug**  

6.1.0 is working fine.

6.2.0 spams the log with:
`
[DevID_3/inputRegs/30781] Calculation of a scaleFactor which is based on another scaleFactor seems strange. Please check the config for address 30781 !

modbus.0
2024-04-12 16:23:48.787	warn	[DevID_3/inputRegs/30779] Calculation of a scaleFactor which is based on another scaleFactor seems strange. Please check the config for address 30779 !
`

**Expected behavior**  

6.2.0 should work as 6.1.0

**Screenshots & Logfiles**  

Config:
[image]

Norb1204, 2024-04-13:
I see the same behavior with version 6.2.0.

Steff42, 2024-04-14:
Here the same.
I have several formulas like this one `x*Math.pow(10, sf['40082'])` in my modbus registers and now every one is marked as warning.
In the ioBroker Forum there is a thread
[https://forum.iobroker.net/topic/74076/gel%C3%B6st-warn-nach-update-auf-modbus-6-2-0](url)
with a solution. But this works not for me.
Mayby i have to try other calculations in my formulas.

greatfireball, 2024-04-14:
Same here... I am using a Solaredge Inverter.

tops4u, 2024-04-15:
Same here. Using 2x SMA. Part of my Config:

[image]

Strobelix80, 2024-04-15:
Same here with formular like `x*Math.pow(10, sf['40082'])`

PLCHome, 2024-04-15:
[image]

Unfortunately I removed two lines too many.

If you insert the two marked lines the error is gone.
The reason was that I wanted to allow strings as SF. This works alternately. Unfortunately he added all the lines. Unfortunately it failed me during the tests.

I'll do a PR right away. Sorry for that.

Maybe someone else can test whether the error has really gone away for them.

PLCHome, 2024-04-15:
The corrected version can be found here:
https://github.com/PLCHome/ioBroker.modbus/blob/master/lib/master.js

PLCHome, 2024-04-15:
@Steff42

> Here the same. I have several formulas like this one `x*Math.pow(10, sf['40082'])` in my modbus registers and now every one is marked as warning. In the ioBroker Forum there is a thread [...]

If you know where the error comes from, I can understand why it didn't work for you.

Steff42, 2024-04-15:
@PLCHome,
thank you for your reply, but my Javascript skills are not yet so good that I understand the reason for this.
I try your corrected version.

> Sorry for that.

No problem, we are all human!

Thx Steffen

Steff42, 2024-04-15:
@PLCHome,
corrected version works fine!

Thx Steffen

PLCHome, 2024-04-15:
Thanx for testing.

tops4u, 2024-04-16:
@PLCHome your file works for me. Thanks!

Apollon77, 2024-04-16:
6.2.1 on it's way

Apollon77, 2024-04-16:
Thanks @PLCHome  for the fix!

closed 2024-04-16 by Apollon77