ioBroker/ioBroker.modbus #502: Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 96

miltonkeans, 2024-04-18:
**Describe the bug**  
Get some error messages 👍 
<html>
<body>

Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 96
--
modbus.0 | 12413 | 2024-04-18 11:28:02.978 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 88
modbus.0 | 12413 | 2024-04-18 11:28:02.977 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 78
modbus.0 | 12413 | 2024-04-18 11:28:02.977 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 76
modbus.0 | 12413 | 2024-04-18 11:28:02.977 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 46
modbus.0 | 12413 | 2024-04-18 11:28:02.976 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 44
modbus.0 | 12413 | 2024-04-18 11:28:02.976 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 42
modbus.0 | 12413 | 2024-04-18 11:28:02.976 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 40
modbus.0 | 12413 | 2024-04-18 11:28:02.976 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 38
modbus.0 | 12413 | 2024-04-18 11:28:02.975 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 36
modbus.0 | 12413 | 2024-04-18 11:28:02.975 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 34
modbus.0 | 12413 | 2024-04-18 11:28:02.975 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 32
modbus.0 | 12413 | 2024-04-18 11:28:02.975 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 30
modbus.0 | 12413 | 2024-04-18 11:28:02.974 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 28
modbus.0 | 12413 | 2024-04-18 11:28:02.974 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 26
modbus.0 | 12413 | 2024-04-18 11:28:02.974 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 24
modbus.0 | 12413 | 2024-04-18 11:28:02.974 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 22
modbus.0 | 12413 | 2024-04-18 11:28:02.973 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 20
modbus.0 | 12413 | 2024-04-18 11:28:02.973 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 18
modbus.0 | 12413 | 2024-04-18 11:28:02.973 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 16
modbus.0 | 12413 | 2024-04-18 11:28:02.972 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 14
modbus.0 | 12413 | 2024-04-18 11:28:02.972 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 12
modbus.0 | 12413 | 2024-04-18 11:28:02.972 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 10
modbus.0 | 12413 | 2024-04-18 11:28:02.971 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 8
modbus.0 | 12413 | 2024-04-18 11:28:02.970 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 6
modbus.0 | 12413 | 2024-04-18 11:28:02.968 | error | Can not set value: The value of "offset" is out of range. It must be >= 0 and <= 2. Received 4

</body>
</html>

Using a RS485 Adapter to connect my  Q Cell PV System.

Input Registers are : 

_address	name	description	unit	type	len	factor	offset	formula	role	room	cw	isScale
3	PV Spannung 1		V	uint16be	1	0,1	0		value		true	false
4	PV Spannung 2		V	uint16be	1	0,1	0		value		true	false
5	PV Strom 1		A	uint16be	1	0,1	0		value		true	false
6	PV Strom 2			uint16be	1	0,1	0		value		true	false
8	Temperatur		°C	int16be	1	1	0		value		true	false
9	aktueller Mode	0=Waiting 1=Checking 2=Normal 3=Fault 4=Permanent Fault 5=Update 6=Off-grid waiting 7=Off-grid 8=Self Testing 9=Idle 10=Standby		uint16be	1	1	0		value		true	false
10	PV Leistung 1		W	uint16be	1	1	0		level		true	false
11	PV Leistung 2		W	uint16be	1	1	0		level		true	false
20	Ladespannung		V	int16be	1	0,1	0		level		true	false
21	Ladestrom		A	int16be	1	0,1	0		level		true	false
22	Ladeleistung		W	int16be	1	1	0		level		true	false
23	BMS Verbindung	0=getrennt, 1=verbunden		uint16be	1	1	0		level		true	false
24	Temperatur Batterie		°C	int16be	1	1	0		level		true	false
25	Status Ladung	0=entladen, 1=laden, 2=stop		uint16be	1	1	0		level		true	false
26	Netzstatus	0=Netz, 1=kein Netz		uint16be	1	1	0		level		true	false
28	Kapazitaet Batterie		%	uint16be	1	1	0		level		true	false
29	OutputEnergy Charge LSB		kWh	uint16be	1	0,1	0		level		true	false
30	OutpurEnergy Charge MSB		kWh	uint16be	1	0,1	0		level		true	false
32	OutputEnergy Charge today		kWh	uint16be	1	0,1	0		level		true	false
33	InputEnergy Charge LSB		kWh	uint16be	1	0,1	0		level		true	false
34	InputEnergy Charge MSB		kWh	uint16be	1	0,1	0		level		true	false
35	InputEnergy Charge today		kWh	uint16be	1	0,1	0		level		true	false
36	BMS ChargeMaxCurrent		A	uint16be	1	0,1	0		level		true	false
37	BMS DischargeMax Current		A	uint16be	1	0,1	0		level		true	false
38	BMS Batterie Kapazitaet		Wh	uint16be	1	1	0		level		true	false
70	Bezug/Ertrag LSB		W	int16be	1	1	0		level		true	false
71	Bezug/Ertrag MSB		W	int16be	1	1	0		level		true	false
72	Gesamtertrag		kWh	uint32sw	2	0,01	0		level		true	false
74	Gesamtbezug		kWh	uint32sw	2	0,01	0		level		true	false
[... 49 lines cut ...]
201	Zaehler Spannung		V	uint16be	1	0,1	0		level		true	false
206	Zaehler Strom		A	int16be	1	0,1	0		level		true	false
256	Modbus Zustand	0=fernbedienung deaktiviert, 1=fernbedienung aktiviert, 2=elektrische Mengenkontrolle aktiviert, 3=Zielkontrolle aktiviert		uint16be	1	1	0		level		true	false

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 6.1.0
 - JS-Controller version: 5.0.19
 - Node version: 18.19.0
 - Operating system: docker

**Additional context**  
Add any other context about the problem here.

Apollon77, 2024-04-18:
which value you try to set? Please post a debug log of such an action too a- and please use most current (Beta) version for this. Thank you

miltonkeans, 2024-04-19:
So far as i understood it is just reading input registers by using a RS485 Adapter from my Q Cell PV system. 
I will come back to you when i did a debug log with the Beta Version.
Thank you

miltonkeans, 2024-04-19:
How do i get the beta version for iobroker ? Which version is it ?

Apollon77, 2024-04-19:
Beta is 6.2.1 - see Readme on top.

Install with Admin - Expertmode - custom ionstall (GitHub icon) - tab "npm" - choose adapter.

For me it sounds like a miss configuration or a read error ... he expects to read starting offset 88 but the received data only have 2 bytes

miltonkeans, 2024-04-20:
Hi 
here is a log with some errors in
Kind Regards
[iobroker.2024-04-20.log](https://github.com/ioBroker/ioBroker.modbus/files/15047657/iobroker.2024-04-20.log)

Apollon77, 2024-04-20:
closed because?

miltonkeans, 2024-04-20:
Sorry. I closed the issue by an mistake

miltonkeans, 2024-04-22:
Sorry my fault. I have reopen the issue.

 

Regards

Apollon77, 2024-05-25:
ok for me it seems that some of your registter definitions are wrongm, an error and no data are returned but he tries to parse that 0 byte response which leads to these errors. Please check the registers