ioBroker/ioBroker.modbus #505: Please add client ID in timeout log message

HansBorjes, 2024-05-01:
**Is your feature request related to a problem? Please describe.**
From time to time I find situations like this in my log files.

modbus.0
2024-05-01 09:38:54.986	warn	Poll error count: 1 code: {"err":"timeout","timeout":5000}

modbus.0
2024-05-01 09:38:54.981	error	Client in error state.

modbus.0
2024-05-01 09:38:54.980	error	Request timed out.

modbus.0
2024-05-01 09:38:54.977	warn	Error: undefined

**Describe the solution you'd like**
Please add the ID of the device not responding to the log file.

**Describe alternatives you've considered**
I have hacked the code myself to add this info. Obviously on the next SW update my patch is gone. :-)

**Additional context**
I suspect that one device in my network has hardware issues and want to isolate that quickly.

Apollon77, 2024-05-25:
Please check version I publish in some minutes

HansBorjes, 2024-05-25:
Thanks a lot for giving it a try! I installed your version. Now it may take some time (1-2 weeks maybe) for the errors to appear again. I will have a look in the log files regularly now.

starting. Version 6.2.3 (non-npm: ioBroker/ioBroker.modbus) in /opt/iobroker/node_modules/iobroker.modbus, node: v18.20.3, js-controller: 5.0.19

HansBorjes, 2024-05-31:
@Apollon77 your version seems to do what it is supposed to do. Thanks!

Can you maybe add a hint like "client ID"?

Interestingly this time it is not caused by a specific client. It seems random. Maybe my bus timing is too challenging.

modbus.0 2024-05-31 16:59:57.200	warn	[11] Poll error count: 1 code: {"err":"timeout","timeout":5000}
modbus.0 2024-05-31 16:59:57.197	error	Client in error state.
modbus.0 2024-05-31 16:59:57.197	error	Request timed out.
modbus.0 2024-05-31 16:59:57.196	warn	Error: undefined