ioBroker/ioBroker.modbus #508: Implement Float (Little Endian Byte Swap)

dieterlind, 2024-05-06:
**Is your feature request related to a problem? Please describe.**
To be able to read all input registers for my IDM heat pump it is required to have the type Float (Little Endian Byte Swap) available for selection.

**Describe the solution you'd like**
Implement type Float (Little Endian Byte Swap)

**Describe alternatives you've considered**
No alternative known

PLCHome, 2024-05-06:
floatbe - Float (Big Endian): AABBCCDD => AABBCCDD
floatle - Float (Little Endian): AABBCCDD => DDCCBBAA
floatsw - Float (Big Endian Word Swap): AABBCCDD => CCDDAABB
floatsb - Float (Big Endian Byte Swap): AABBCCDD => DDCCBBAA

Maybe the name was just chosen poorly
if floatle - Float (Little Endian): AABBCCDD => DDCCBBAA
Then by Byte Swap CCDDAABB so it must be floatsw - Float (Big Endian Word Swap) the same. Isn't it?

Or what is the expected result from the bytes in hex representation with the identifier AA BB CC DD of each byte sendet in  order A to D?

It's actually funny that in the documentation floatsb is the same as floatle.

PLCHome, 2024-05-06:
>>When I read register 1000 or 1002 it works, when I try to read 1008 it doesn't work.
Ok, der Datentyp scheint es aber nicht zu sein. Wenn diese Werte korrekt sind.

>>Error when reading 1008:
I think it's more of an alignment problem. He cannot access the address.

dieterlind, 2024-05-09:
The strange thing is that 1000 and 1002 are correct with Float (Big Endian Word Swap) in ioBroker, but the values are wrong in Modbus Poll when setting to Big Endian Byte Swap (no word swap available). They are just right there when using Little Endian byte Swap.

But how to fix the issue in iob-modbus if the alignment seems to be right in Modbus Poll?

dieterlind, 2024-05-15:
Any ideas?