ioBroker/ioBroker.modbus #514: Speicherleck

edyso, 2024-05-30:
**Describe the bug**  
Speicherverbrauch der Modbus-Instanz wächst pro Tag um ca. 70-80MB

**To Reproduce**  
nach jebem Restart startet der Speicherverbrauch bei 75MB und wächst täglich.

**Screenshots & Logfiles** 
Über die Debug-Console scheint ein Timer-Array (_timers) für den Speicherverbrauch verantworlich zu sein.
[image]
[image]

**Versions:**  
 - Adapter version: 6.1.0 auch 6.2.3
 - JS-Controller version: 5.0.19
 - Node version: 18.20.2
 - Operating system: Debian

Apollon77, 2024-05-31:
Good catch ...

Apollon77, 2024-05-31:
@edyso Can oyu please try GitHub version and verify that the issue is fixed with this?

closed 2024-05-31 by Apollon77

edyso, 2024-05-31:
I'm testing...I'll report tomorrow.
For your information, my pooling-time is 100ms, I've a push button

Apollon77, 2024-05-31:
Thank you, looking forward to your results

edyso, 2024-06-01:
I'm sorry, but the _timers array is stil growing, after round about 8 hours the length of array is 975573 (number after @).
Your fix affects the lib/master.js so I checked if the file on my server is the same like on github.

Apollon77, 2024-06-01:
And you made sure that the instance had restarted after installing the update from GitHub? Because the version did not change it could be that there was no auto restart? or did you restaretd manually?

edyso, 2024-06-01:
I'm sure that the changes made by you are live, first I was confused about the version, than I checked the file that the modifications are included and I inserted a log row. After restart I get the log entry.

edyso, 2024-10-17:
The bug is still present in the current verion 6.3.2.
My workaround is to restart the adapter every 24h or to set a memory limit, this ends with log entries, maybe there are also sentry entries...

commit 42ea88d5 (2024-05-31): * (Apollon77) Fix Timeout management to prevent leaking memory, fixes #514