ioBroker/ioBroker.modbus #515: Two RPI Adapter as  ModbusMaster

Syil237, 2024-05-31:
I have two RPIs in the network and an energy meter Siemens PAC3200. When I access the energy meter with one of them, I can read the meter. When I then activate the second adapter, the first one can read but the second one cannot. It doesn't matter which one is activated first. Both adapters are set as Master and on TCP.
[image]
[image]

I want to be able to access the energy meter with both RPIs simultaneously.

Apollon77, 2024-05-31:
What does the log say of the instance that do not work? Is the energy meter Siemens PAC3200 capable to handle two concurrent TCP connections?

sakasitadi, 2024-06-03:
ok, its solved. 
[image]

Apollon77, 2024-06-04:
So the device was only supporting one connection?

closed 2024-06-04 by Apollon77