ioBroker/ioBroker.modbus #539: Instance does not start after upgrading from v6.1.0 to 6.2.3

MPTrixi, 2024-07-12:
**Describe the bug**  
When updating from v6.1.0 to v6.2.3 I get for one out of three running instances errors in log (adapter stays yellow):
from 1 up to 13

```
modbus.2 | 2024-07-12 14:43:39.419 | error | Reconnect did not help, restart adapter
modbus.2 | 2024-07-12 14:43:39.418 | warn | [1] Poll error count: 13 code: {"errorCode":130,"exceptionCode":0}
```

The device is a Protoss-PW11 RS485 to Wi-Fi

**To Reproduce**  
Steps to reproduce the behavior:
happens immediatly after updating the adapter at the attempt to restart instance
Downgrading to 6.1.0 solved the issue

**Expected behavior**  
instance should run after update of adapter

**Screenshots & Logfiles**  
`
modbus.2 2024-07-12 14:43:39.419	error	Reconnect did not help, restart adapter
modbus.2 2024-07-12 14:43:39.418	warn	[1] Poll error count: 13 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:43:39.344	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:42:39.339	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:42:39.287	warn	[1] Poll error count: 12 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:42:39.210	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:41:39.204	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:41:39.153	warn	[1] Poll error count: 11 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:41:39.078	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:40:39.074	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:40:39.023	warn	[1] Poll error count: 10 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:40:38.942	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:39:38.936	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:39:38.883	warn	[1] Poll error count: 9 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:39:38.806	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:38:38.799	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:38:38.747	warn	[1] Poll error count: 8 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:38:38.671	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:37:38.662	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:37:38.609	warn	[1] Poll error count: 7 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:37:38.530	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:36:38.522	info	Disconnected from slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:36:38.469	warn	[1] Poll error count: 6 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:36:33.332	warn	[1] Poll error count: 5 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:36:28.205	warn	[1] Poll error count: 4 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:36:23.073	warn	[1] Poll error count: 3 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:36:17.943	warn	[1] Poll error count: 2 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:36:12.811	warn	[1] Poll error count: 1 code: {"errorCode":130,"exceptionCode":0}
modbus.2 2024-07-12 14:36:12.718	info	Connected to slave xxx.xxx.xxx.xxx
modbus.2 2024-07-12 14:36:06.273	info	starting. Version 6.2.3 in /opt/iobroker/node_modules/iobroker.modbus, node: v20.14.0, js-controller: 5.0.19`

[image]

**Versions:**  
 - Adapter version: 6.2.3
 - JS-Controller version: 5.0.19 
 - Node version: 20.14.0 
 - Operating system: Debian (Raspberry)

**Additional context**  
Add any other context about the problem here.

Apollon77, 2024-07-12:
And you are sure that it is not the devcie which has issues?

MPTrixi, 2024-07-12:
when I go back to 6.1.0 it works

dolomiti70, 2024-07-14:
I had the same problem. I was wondering why it works with my testsytem and not with my main system. In my testsytem im not using "discrete inputs". So i added an discrete input and i had the same error as described above.
When i marked "Do not align addresses to 16 bits" everything is working fine.

Dirk-Ma, 2024-07-18:
I have the samoe problem with Wago. I have no "discete inputs". If I marked "Do not align addresses to 16 bits" everything is working fine.

GermanBluefox, 2024-08-28:
The error could only come because of it: https://github.com/ioBroker/ioBroker.modbus/pull/510

GermanBluefox, 2024-08-28:
Please try 6.3.3 and if the version works without "Do not align addresses to 16 bits" too. If it works, so the PR https://github.com/ioBroker/ioBroker.modbus/pull/510 must be checked again

GermanBluefox, 2024-08-29:
> Hello, where can I get 6.3.3? With 6.3.0 it works with and without "Do not align addresses to 16 bits".

I meant 6.3.0

GermanBluefox, 2024-08-29:
@Q7Jensen https://github.com/ioBroker/ioBroker.modbus/pull/510 (lets discuss in this issue)

GermanBluefox, 2024-08-29:
@Dirk-Ma I think I could find the error. Please test the 6.3.1

Q7Jensen, 2024-08-29:
@GermanBluefox 
I looked at my PullRequest again and tested it and could not see the problem. Please lets discuss this issue promptly and preferably in German.

Q7Jensen, 2024-08-30:
@GermanBluefox 
You're absolutely right.
It should read    result.blocks[b].count += (oldstart - result.blocks[b].start);

MPTrixi, 2024-08-30:
> @Dirk-Ma I think I could find the error. Please test the 6.3.1

@GermanBluefox Installed 6.3.2 from GitHub and instances are coming up green, I get data and can trigger the devices. Looks good now for me