ioBroker/ioBroker.modbus #563: Synchronisation of two instances

dannyocean1, 2024-09-25:
Hi,

can somebody help me with this question? I habe two modbus instances running. Each instance is collecting every 20 seconds data of input registers of a Sungrow pv inverter. Is it possible to synchronize the polling of both instances? At the moment there is a difference of some seconds between the instances.

Thanks for helping

Apollon77, 2024-09-25:
I am not sure how this should work ... you will also not be really able to make sure that devices answer at the same speed and and and what exactly is the goal of this?

dannyocean1, 2024-09-26:
I want to calculate the electric energy flow. The sum of energy production, consumtion and grid feed should be zero. That is not the case in the moment maybe due to the time differences. Of course you are right that the answering speed may be different at least. But I think nevertheless it should help when the requests are synchronized.

GermanBluefox, 2025-09-15:
It is impossible. Normally I subscribe to both states in js script and calculate on the change of any state with the known last values the desired parameters

closed 2025-09-15 by GermanBluefox