ioBroker/ioBroker.modbus #573: Memory leak, closed issue #514

edyso, 2024-10-17:
The bug is still present in the current version 6.3.2.
My workaround is to restart the adapter every 24h or to set a memory limit, this ends with log entries, maybe there are also sentry entries...

Old, closed issue #514

tobi119, 2025-03-25:
Seams to be the same issu at me.
A few days after restart an error message appears in the log that repeats every few hours. Restarting the Modbus-Instance resets the error for some days.
 

[image]

I noticed that the instance's memory usage was growing. 
So I triggerd the data point "system.adapter.modbus.6.memHeapTotal"  that shows the growth over the time.

Workaround: If memory usage of the heap exceeds 80MB, the instance is restarted.

Picture: growth of memory usage (heap) over 2 days after restart
[image]

sansonifabio, 2025-03-26:
... auch ich habe die gleichen Probbleme.

Xander0823, 2025-04-19:
Hier ebenfalls, mein Raspi3 stürzt nach 2 Tagen Betrieb mit dem ModBus-Adapter wegen Speichermangel ab :-(

mpl1337, 2025-05-29:
So sieht es bei mir aus

[image]

<br />
<br />
Heute kam folgender fehler:
<br />
<br />

<html>
<body>

modbus.2 | 2025-05-29 20:55:30.010 | error | Can not set value: Attempt to access memory outside buffer bounds
<br />
modbus.2 | 2025-05-29 20:55:30.007 | error | Can not set value: Attempt to access memory outside buffer bounds

</body>
</html>