ioBroker/ioBroker.modbus #595: Several times a day "Socket Error", "Client in error state." and "Request timed out."

exaware, 2024-11-28:
I have two instances of the Modbus adapter running. One adapter communicates with my Sungrow inverter, and the other adapter with a heating element "my-PV AC ELWA 2". Basically, everything works, and I receive all the values. However, the problem is that the following errors appear daily in the ioBroker log. The adapters continue running, and values are still received. I have already experimented with the timings in the settings, but I cannot eliminate the errors. I have also reduced the register lines to rule out faulty registers, but unfortunately, without success.

[image]

[image]

tobi119, 2025-03-25:
The same as #573? Look at the memory usage of the adapter

sansonifabio, 2025-03-26:
... auch ich habe die selben Probbleme.

exaware, 2025-10-17:
Any news on this?

mcm1957, 2025-10-17:
Please check if this error still occures with current latest release (7.0.5)

exaware, 2025-11-15:
Ich habe auch schon viel mit den Einstellungen meines Heizstabes gespielt, leider ohne Erfolg. Ich bekomme den Fehler ziemlich oft. Habe das Log-Level nun auf Alles gestellt. Werde morgen hier Logs posten. 
Liebe Grüße Cetin.

exaware, 2025-11-16:
@Apollon77 kann es sein, dass der beschrieben Fehler etwas mit den Log-Levels zusammenhängt bzw. ein Fake Fehler ist? Ich habe zwei Modbus-Instanzen, einmal für meinen Heizstab und einmal für meinen Wechselrichter. Bei beiden Instanzen habe ich die hier besprochenen Fehler bekommen. Gestern habe ich das Log-Level beider Instanzen auf "Alles" (Silly) gestellt und komischerweise ist der Fehler bisher bei beiden Instanzen nicht aufgetreten.

Ich warte mal ab, sobald ich den Fehler habe, werde ich diesen hier posten.

exaware, 2025-11-18:
Ich denke ich habe ein brauchbares Log. Ich denke der Fehler passiert beim Schließen der Verbindung. 
Vielleicht dauert das Schleißen in manchen Fällen zu lange.

**Hier ein Durchgang ohne Fehler:**
[OK.txt](https://github.com/user-attachments/files/23614043/OK.txt)

**Und hier ein Durchgang mit Socket-Error:**
[NOK.txt](https://github.com/user-attachments/files/23614044/NOK.txt)

nobl, 2026-08-17:
Ich kann ein sehr ähnliches Verhalten mit `ioBroker.modbus` 9.1.0 reproduzieren. Mein Aufbau unterscheidet sich allerdings etwas vom ursprünglich beschriebenen: Ich verwende eine einzige Modbus-Adapterinstanz im TCP-Master-Modus mit zwei Geräte-IDs.

## Aufbau

* zwei Eastron SDM630 am selben RS485-Bus
* Geräte-IDs 1 und 2
* RS485 als durchgehender Bus verdrahtet
* 120-Ω-Abschluss an beiden Busenden
* zusätzlich gemeinsame GND-Verbindung
* Waveshare RS485 TO POE ETH (B) als Modbus-TCP/RTU-Gateway
* Waveshare-Firmware V1.523
* Netzwerkverbindung über PoE
* Modbus TCP auf Port 502
* serielle Einstellungen: 9600 Baud, 8 Datenbits, keine Parität, 1 Stoppbit
* Waveshare-Protokoll: „Modbus TCP to RTU“
* Multi-Host aktiviert
* Instruction Timeout: 1024 ms
* RS485 Conflict Time Gap: 100 ms

Das Waveshare ist während der Fehler weiterhin ohne Unterbrechung per Ping erreichbar. Ein parallel laufender Dauer-Ping zeigt keine Paketverluste und durchgehend Antwortzeiten unter 1 ms. Ein Ausfall der Ethernetverbindung, des Switchports oder der PoE-Versorgung ist daher sehr unwahrscheinlich.

## ioBroker-Konfiguration

* `ioBroker.modbus` 9.1.0
* Node.js 22.23.2
* js-controller 7.2.2
* TCP Master
* mehrere Geräte-IDs aktiviert
* Datenabfrageintervall: 5000 ms
* Lese-Timeout: 2000 ms
* Leseintervall zwischen zwei Anforderungen: 200 ms
* maximale Leseanforderungslänge für Float: 2 Register
* maximale Adresslücke zum Zusammenfassen: 0

Es werden nur wenige Input-Register gelesen:

Geräte-ID 1:

* Adresse 52, Länge 2
* Adresse 72, Länge 2
* Adresse 74, Länge 2

Geräte-ID 2:

* Adresse 52, Länge 2
* Adresse 72, Länge 2

Alle Werte sind Float Big Endian.

## Normaler Ablauf

Im fehlerfreien Zustand werden die Anforderungen sauber nacheinander ausgeführt. Die Antwortzeit liegt normalerweise lediglich bei etwa 27–30 ms:

```text
22:43:56.904  [DevID_1] Poll start
22:43:56.905  [DevID_1/inputRegs] Poll address 52 - 2 registers
22:43:56.933  [DevID_1/inputRegs] Poll address 52 DONE
22:43:57.135  [DevID_1/inputRegs] Poll address 72 - 2 registers
22:43:57.162  [DevID_1/inputRegs] Poll address 72 DONE
22:43:57.364  [DevID_1/inputRegs] Poll address 74 - 2 registers
22:43:57.392  [DevID_1/inputRegs] Poll address 74 DONE

22:43:58.094  [DevID_2] Poll start
22:43:58.095  [DevID_2/inputRegs] Poll address 52 - 2 registers
22:43:58.133  [DevID_2/inputRegs] Poll address 52 DONE
22:43:58.334  [DevID_2/inputRegs] Poll address 72 - 2 registers
22:43:58.362  [DevID_2/inputRegs] Poll address 72 DONE
```

Es gibt in diesem Zustand keine überlappenden Anforderungen. Auch die Abstände zwischen den einzelnen Anforderungen und den Geräte-IDs sind deutlich größer als die üblichen Antwortzeiten.
[... 163 lines cut ...]

* Geräte-ID 1: Register 52, 72 und 74
* Geräte-ID 2: Register 52 und 72
* die Antwort auf ID 1 / Register 72 wird einmal absichtlich unterdrückt

Mit dem unveränderten Upstream-Code schlägt der Test fehl: Nach dem Reconnect wird zuerst die alte 74er-Anfrage übertragen und der Zyklus für Geräte-ID 2 läuft doppelt.

Mit dem Patch wird die alte Poll-Kette beim Timeout vollständig beendet. Nach dem Reconnect beginnt genau ein neuer Zyklus.

Der Patch wurde außerdem auf dem realen System getestet. Der erste Timeout trat dort weiterhin auf, aber die anschließende Verarbeitung war nun sauber:

* Timeout bei ID 1 / Register 72
* Socket geschlossen und FIFO bereinigt
* keine weitere alte Registeranfrage
* Reconnect
* genau ein neuer Zyklus: ID 1 mit 52 → 72 → 74
* danach ID 2 mit 52 → 72
* keine überlappenden oder doppelten Poll-Schleifen

Die vollständige Testsuite der Bibliothek läuft mit dem Patch durch (`65 passing`); TypeScript-Prüfung, ESLint und Build waren ebenfalls erfolgreich. Auch die Tests des einbindenden ioBroker-Adapters einschließlich Integrationstest liefen erfolgreich.

Wichtig: Der Patch behebt nicht zwingend die Ursache des zuerst verlorenen Modbus-Replies. Er behebt den zusätzlichen Fehler in der Reconnect-Behandlung, durch den verwaiste Anfragen und parallele Poll-Zyklen entstehen konnten. Es ist denkbar, dass gerade diese Kaskade langfristig zu weiteren Kommunikationsproblemen oder einem festgefahrenen Bus beiträgt.

Bei mir hatte das ganze zumindest auf Dauer dazu geführt, dass die Appliances einfach gar nicht mehr reagiert haben.
Unangenehm, wenn sichs dabei um Stromzähler handelt, die man nur mit drei Sicherungsautomaten gleichzeitig resetten kann :)