ioBroker/ioBroker.modbus #615: Error in Handler

mpl1337, 2025-01-05:
Woher kommt diese Warnung?

[image]

[image]

[image]

```
Plattform
linux
Betriebssystem
linux
Architektur
x64
CPUs
4
Geschwindigkeit
3600 MHz
Modell
Intel(R) Core(TM) i3-8109U CPU @ 3.00GHz
RAM
5.16 GB
System-Betriebszeit
6 T. 18:15:10
Node.js
v20.18.0 (Es gibt eine neuere Version: v20.18.1 - Empfohlene Version v18.20.5)
time
1736086500802
timeOffset
-60
NPM
10.9.2
Anzahl der Adapter
554
Datenträgergröße
28 GB
freier Festplattenspeicher
20.25 GB
Aktive Instanzen
35
Pfad
/opt/iobroker/
Betriebszeit
1 T. 17:50:59
Hostname
iobroker
```

Apollon77, 2025-01-05:
Enable debug log and potentially the message level debug in adapter settings. then you should see the data. It seems that there is an data parsing error

mpl1337, 2025-01-06:
Ich habe es bereits versucht, jedoch wird mein Log so voll, dass sich das ganze System aufhängt sobald ich diesen öffnen will.

Apollon77, 2025-01-06:
Wie versuchst Du Ihn denn zu öffnen?